interactive-matter/aJson #66: Updating a boolean value

lasselukkari, 2015-01-05:
I was wondering why the boolean type has been split to aJson_False and aJson_True?

Because of this updating a boolean value may work in an unexpected way.

For example if I update a boolean value directly like this
```cpp
aJson.getObjectItem(user, "registered")->valuebool = false; // was true before
```
and the print the obeject it still prints out as "true" because the printValue function only checks for the type and not the actual value.

I have created a fork where the values have been combined to a single type, but I didn't create a pull request yet because I think it's possible that I have just misunderstood something obvious.

interactive-matter, 2015-01-05:
hmm, checking the my code at this point - could be rubbish - do not get my head around the fact Why there is aJson_False and aJson_True … could have been an oversimplification.
I think it is better to solve the root cause, isn't it?
like removing aJson_False and aJson_True for aJson_Bool or so - but that would be a bigger API change …

lasselukkari, 2015-01-05:
Do you want me to create a pull request? I have the changes ready here.

interactive-matter, 2015-01-05:
Excellent! yes, please

closed 2015-01-05 by interactive-matter