interactive-matter/aJson #68: Dictionary support

lasselukkari, 2015-01-05:
Often the key names are repeated in many different objects. For example you have an array of objects where each object contains the same keys. Currently all the key values point to separate strings when they could be the same string in the memory.

I have created a fork where the user can supply a dictionary in form of string array:
the changes can be seen here: https://github.com/lasselukkari/aJson/commit/498fac54a60dc981fa57ea2134d407624ffa0d04

Is this something you think would be usefull to others too?

interactive-matter, 2015-01-05:
yes it is - extremely! I think this makes much more sense for this son implementation - excellent

interactive-matter, 2015-01-05:
(= looking forward to the pull request)

lasselukkari, 2015-01-05:
I will clean it up and make the pull request tomorrow. 

Currently the lenght of the dictionary is determided with last value being a NULL. Do you prefer this or should the user supply the dictionary length as a second parameter?

interactive-matter, 2015-01-05:
btw - nice idea with your WOT - but awful lot of work! ;) wanted to do it myself, got distracted

lasselukkari, 2015-01-05:
Thanks! I was travelling around Asia for three months and had lots of spare time while moving from place to another.

closed 2015-12-30 by lasselukkari