interactive-matter/aJson #70: aJson.parse changing value of double

merrillii, 2015-04-08:
I have a 5 digit precision double that I'm trying to parse and it seems to get munged by the parse method.  The following test code shows the problem.  The output will be:

Full string={"laserWavelen":638.12345}                                       
Total json={"laserWavelen":638.12360}
 
The last two digits always seem to get screwed up.  Any ideas how to fix this?

#include <aJSON.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
  char json_string[] = "{\"laserWavelen\":638.12345}\0";
  
  aJsonObject* jsonObject = aJson.parse(json_string);
  
  Serial.print("Full string=");
  Serial.println( json_string );

  Serial.print("Total json=");
  char *p = aJson.print(jsonObject);
  if (p != NULL) {
    Serial.println(p);
    free(p);
  }
  else
    Serial.println("NULL!");
    
  aJson.deleteItem(jsonObject);
  
  delay(5000);
}

nw2s, 2015-04-08:
Doesn't Arduino only support single?

merrillii, 2015-04-08:
You may be correct that I'm running out of bits.  The arduino doc says 32 bits for a double.  If I change the code to make the test number 0.12345 then it seems to work.  So, any ideas for a workaround to getting a higher precision number through JSON?

merrillii, 2015-04-08:
You are right that it's an Arduino precision issue. Even using string functions, the code fails:

  float d = atof("638.12345");
  char floatBuffer[20];
  Serial.print("float test =");
  dtostrf(d, 19, 10, floatBuffer);
  Serial.println(floatBuffer);

outputs:
  float test =     638.1234700000

closed 2015-04-08 by merrillii