interactive-matter/aJson #75: Use of array

sticilface, 2015-09-27:
I have the following Json string
```
{
  "bri" : 14,
  "xy" : [
    0.2404497,
    0.5637349
  ],
  "on" : true
}
```

I can retrieve the bri values using 
```
    if (aJsonObject* briState = aJson.getObjectItem(parsedRoot, "bri")) {
      bri = briState->valueint;
      //bri /= 255; 
    }
```

but i can't seem to work out how to get the parts 1 and 2 of the "xy"  into a struct containing x and y as floats.  could you point me in the right direction?

nw2s, 2015-09-27:
Any way to do this that's not in an issue? Maybe not. If you need some sample parsing code, check out getIntCollectionFromJSON

at https://github.com/nw2s/b/blob/b-1.1.0-unstable/sketches/libraries/nw2s/JSONUtil.cpp

Hint: You need to know that element is an array and then treat it as such. Collections are different from simple values. They have values for methods that normally wouldn't make sense, namely getArraySize and getArrayItem.

-s

sticilface, 2015-09-27:
I did look but nothing jumped out.  
So I've managed to get it to parse the float as a double, using this:
```
    if (aJsonObject* xyState =  aJson.getObjectItem(parsedRoot, "xy")) {

      //aJsonObject* xyColourSize = aJson.getArraySize(xyState);
      aJsonObject* xyColourSpaceX = aJson.getArrayItem(xyState, 0);
      aJsonObject* xyColourSpaceY = aJson.getArrayItem(xyState, 1);

  if (xyColourSpaceX != NULL && xyColourSpaceX != NULL) {

      //xyColourSpace[0] = xyState->json_string;
      //xyColourSpace.y = xyState->aJson_Array;

      Serial.print("ARRAY RECIEVED SIZE = ");
      Serial.println(aJson.getArraySize(xyState));
       Serial.print("X = ");
       Serial.println(xyColourSpaceX->valuefloat);
}
}
```

However, as you can see from the first post there 7 digits after the decimal, and I could really do with using all of them not just 2.  I thought I could parse as a string and then convert it to float, but parsing as string does not work.  

any idea?

nw2s, 2015-09-29:
So you're saying that serial.println is only printing two decimal places? Before blaming the framework parsing, be sure that it's not the println formatting that's truncating the float. I can't see any more of the code that might or might not be contributing, so that's all I could guess.