interactive-matter/aJson #85: Could you confirm this library is abandoned?

q2dg, 2016-08-08:
I'm deciding between https://github.com/bblanchon/ArduinoJson and this one, but it seems pretty abandoned: no answers on issues, last commit from one year ago...Can you confirm this, please?
Thanks

dskvr, 2016-08-09:
I would say yeah, it's abandoned. The Interactive Matter website is down too. I had problems with this library when I was trying to use it a couple years back. I would go with the maintained one :+1:

nw2s, 2016-08-09:
It's a JSON library (see what I did there?) - Does it need continuing
commits? It's simple, lightweight, and works. I've done my best to answer
any questions folks have even though I'm not the original author. Use what
you're comfortable with, but don't be too quick to judge.

q2dg, 2016-08-13:
Thanks. I just wanted to choose the most supported library for my project. I'm sure this library does the job perfectly, but if I ever had a problem, community is important: what annoyes me a lot is the amount of unanswered issues...that's a bad thing. 
Maybe I'm wrong, but it seems https://github.com/bblanchon/ArduinoJson is more updated lately. Thanks.

interactive-matter, 2016-08-16:
Hi, 
as the original author I suggest using  https://github.com/bblanchon/ArduinoJson . The AJson Library was an effort which for me proved in the end not to be worth the effort. JSON decoding on the Arduino is possible - as this and the ArduinoJSON shows. But int the end I doe not use this library much anymore - since binary serial communication is much more appropriate - even though it may be a bit harder. I use https://github.com/thijse/Arduino-CmdMessenger a lot …
Sorry for the inconvenience

q2dg, 2016-08-19:
Thanks a lot for being so sincere and clear, @interactive-matter 
I'll close the issue, then

closed 2016-08-19 by q2dg