interactive-matter/aJson #89: int flush() in aJsonStream overrides void flush() in Print

ryanneve, 2017-08-14:
It looks like aJsonStream.flush() returns an int, but is trying to override Print.flush() which returns void.

I'm getting compile errors with arduino 1.8.3:
```
In file included from C:\...\Energy_JSON.h:3:0,
                 from C:\...\Energy_JSON.cpp:1:
D:\Development\Arduino\libraries\aJson/aJSON.h:92:6: error: conflicting return type specified for 'virtual int aJsonStream::flush()'
  int flush();
      ^
In file included from D:\Development\Arduino\libraries\aJson/aJSON.h:28:0,
                 from C:\...\Energy_JSON.h:3,
                 from C:\...\Energy_JSON.cpp:1:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Print.h:90:18: error:   overriding 'virtual void Print::flush()'
     virtual void flush() { /* Empty implementation for backward compatibility */ }
```
Changing aJsonStream.flush() so it doesn't return anything fixes the problem, but not sure if it might create other problems.

MopheusDG, 2018-02-02:
Same problem here, can't compile using latest IDE and ESP8266 latest version too. Same error.

bltDefender, 2018-05-13:
same to me here. I commented out the corresponding flush method (in cpp and in h) to make my build work again.

mjhammel, 2018-07-10:
FWIW, I created a patch for this.  I also checked the aJson code and no one is calling flush so it shouldn't hurt the library.  Any application code checking the return code of flush might be affected but it's an easy fix - don't check the return code (it doesn't tell you anything useful anyway).  

Patch can be found here: https://gitlab.com/xarduino/lightsw/blob/master/patch/ajson-void-flush.patch

kingsjl, 2018-08-02:
Arduino:1.8.5 (Windows 10), 开发板："Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from D:\360瀹夊叏娴忚鍣ㄤ笅杞絓aJson-master (1)\aJson-master\Examples\Json_Serial\Json_Serial.ino:19:0:

aJSON.h:92: error: conflicting return type specified for 'virtual int aJsonStream::flush()'

  int flush();

      ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Stream.h:26:0,

                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:29,

                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:232,

                 from sketch\Json_Serial.ino.cpp:1:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Print.h:90:18: error:   overriding 'virtual void Print::flush()'

     virtual void flush() { /* Empty implementation for backward compatibility */ }

                  ^

exit status 1
conflicting return type specified for 'virtual int aJsonStream::flush()'

在文件 -> 首选项开启
“编译过程中显示详细输出”选项
这份报告会包含更多信息。